yujinrobot/kobuki_desktop #41: Kobuki Gazebo5 plugin is not working

jvgomez, 2015-07-15:
It moves very weirdly. After looking for it a lot, I found this commit in a fork:

https://github.com/mayfieldrobotics/kobuki_desktop/commit/4246c6e4330f39725df3eb39a0b77a7fcc6185b1

That seems to solve it. The changes in Gazebo5 are: https://bitbucket.org/osrf/gazebo/pull-request/1218/change-behavior-of-odejoint-setvelocity/diff 
and point 14 Gazebo 5.0.0: https://bitbucket.org/osrf/gazebo/src/default/Changelog.md

Please solve this issue. Thanks!

bit-pirate, 2015-07-16:
> Please solve this issue. Thanks!

We always appreciate pull requests to solve such issues quickly. :-)

jvgomez, 2015-07-16:
If I knew how to do it properly I would do the PR. However that link I gave I do not know whether is just a workaround or the proper way of doing that.