yujinrobot/kobuki_desktop #44: Compiling error about c++11 support at ARM

xiekuncn, 2016-06-08:
I got an error when compiling the **kobuki_gazebo_plugin** at NVIDIA TK1. such as
`
/usr/include/c++/4.8/bits/c++0x_warning.h:32: error:
      #error This file requires compiler and library support for the ISO C++ 2011
             standard. This support is currently experimental, and must be
             enabled with the -std=c++11 or -std=gnu++11 compiler options.
      #error This file requires compiler and library support for the \
`
and I fixed it when I add following code at ./kobukin_gazebo_plugins/CMakeLists.txt.
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLUGS} -std=c++11")`

It may useful for any other use kobuki at ARM.

jvgomez, 2016-06-08:
it is interesting that this plugin depends on C++11 when the whole ROS does not. Which compiler are you using? Because I think that this problem does not appear when compiling with regular GCC without C++11, does it?

stonier, 2016-06-09:
```
#include <cmath>
#include <cstring>
#include <string>
#include <boost/bind.hpp>
#include <boost/thread.hpp>
#include <boost/shared_ptr.hpp>
#include <gazebo/gazebo.hh>
#include <gazebo/common/common.hh>
#include <gazebo/common/Time.hh>
#include <gazebo/math/gzmath.hh>
#include <gazebo/physics/physics.hh>
#include <gazebo/sensors/sensors.hh>
#include <gazebo_plugins/gazebo_ros_utils.h>
#include <ros/ros.h>
#include <std_msgs/Empty.h>
#include <std_msgs/Float64.h>
#include <sensor_msgs/Imu.h>
#include <sensor_msgs/JointState.h>
#include <nav_msgs/Odometry.h>
#include <geometry_msgs/Twist.h>
#include <geometry_msgs/TransformStamped.h>
#include <tf/transform_broadcaster.h>
#include <tf/LinearMath/Quaternion.h>
#include <kobuki_msgs/MotorPower.h>
#include <kobuki_msgs/CliffEvent.h>
#include <kobuki_msgs/BumperEvent.h>
```

That's a list of all the includes. Nothing esoteric there, and all of what is there doesn't need c++0x/c++11 support.

rohbotics, 2016-07-09:
Are you using Kinetic (or Jade) on the TK1?
Both ship with versions of Gazebo that require C++11.

My PR #45 should fix it (at least for kinetic).