yujinrobot/kobuki_desktop #50: kobuki_gazebo not compiling after gazebo_ros_pkgs cleanup

tfoote, 2017-04-22:
There was a dependency cleanup in gazebo_ros_pkgs https://github.com/ros/rosdistro/pull/14563

There's a known issue in catkin
https://github.com/ros/catkin/issues/856

stonier, 2017-04-22:
@tfoote had a read of all that and managed to reproduce the error locally. 

I don't think I can do anything from here though - if the workaround with a config extras cmake module is the right way for now (as demonstrated by @davetcoleman in [dummy_gazebo_ws](https://github.com/davetcoleman/dummy_gazebo_ws/), then should this be applied in [gazebo_plugins](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/CMakeLists.txt#L143) which still expicitly DEPENDS on GAZEBO?

tfoote, 2017-04-22:
@j-rivero @dirk-thomas Can you comment?

dirk-thomas, 2017-04-23:
I don't have anything to contribute on this. As far as I recall the catkin ticket this was supposed to be addressed within the packages affected somehow. And if a catkin change could be helpful someone wanted to provide a PR for it.

j-rivero, 2017-04-24:
> @j-rivero @dirk-thomas Can you comment?

Dirk is right, we don't count with patching catkin at least in the short term. If @davetcoleman has time to provide a solution that would be great. Otherwise I will revert the change in `gazebo_ros_pkgs`.

stonier, 2017-04-24:
If going forward is important, as far as I can tell,  we could quickly patch `gazebo_plugins` with @davetcoleman's workaround (i.e. shift Gazebo from DEPENDS to CFG_EXTRAS in the package exports) or alternatively wait for [gazebo_dev](https://github.com/ros-simulation/gazebo_ros_pkgs/pull/473/files) to land which does exactly this and depend on that.

tfoote, 2017-04-24:
@j-rivero I think that reverting until there's a clear recommended path forward is the best path for now.

j-rivero, 2017-04-24:
while I clean up the mess and provide a decent alternative, here is the revert so we don't make the new kinetic release to wait: https://github.com/ros/rosdistro/pull/14631. Thanks @tfoote @stonier

davetcoleman, 2017-04-25:
This is pretty annoying - that cleanup has now been reverted twice. It was important to me because I was asked to get the Baxter SDK (that uses Gazebo) to have no build warnings, now that it is reverted it is again incorrect and produces the following output:

```
Warnings   << gazebo_plugins:check /home/dave/ros/current/ws_vicarious/logs/gazebo_plugins/build.check.001.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'gazebo' but neither 'gazebo_INCLUDE_DIRS' nor
  'gazebo_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:88 (catkin_package)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'SDF' but neither 'SDF_INCLUDE_DIRS' nor
  'SDF_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:88 (catkin_package)
```

tfoote, 2017-04-26:
Fixed by rollback. Closing.

closed 2017-04-26 by tfoote

j-rivero, 2017-04-26:
> This is pretty annoying - that cleanup has now been reverted twice. It was important to me because I was asked to get the Baxter SDK (that uses Gazebo) to have no build warnings, now that it is reverted it is again incorrect and produces the following output:

That is mostly my fault for not being to find a good solution, sorry for that Dave. I hope that by merging https://github.com/ros-simulation/gazebo_ros_pkgs/pull/571 the problems is solved for all the parts.