yujinrobot/kobuki_desktop #53: Pyrcc5: not found

wangzhaodong123, 2017-05-18:
I met a problem when i make my space. It stopped here:
[ 98%] Built target kobuki_nodelet
/bin/sh: 1: pyrcc5: not found
make[2]: *** [kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite] 错误 127
make[1]: *** [kobuki_desktop/kobuki_qtestsuite/CMakeFiles/kobuki_qtestsuite.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j1 -l1" failed
Can you tell me the reason and help me solve it? Thank you.

stonier, 2017-05-18:
You already raised this issue in #51?

RobotQZ, 2018-01-26:
I have met the same problem now,anyone can help me? please

Abdul-Mukit, 2019-01-17:
`sudo apt-get install pyqt5-dev-tools`
worked for me.

stonier, 2019-01-21:
What distro are you on?

stonier, 2019-01-21:
If you have version >= 0.5.7 rosdep should nab it for you. Making this a duplicate of #51.

closed 2019-01-21 by stonier