yujinrobot/kobuki_desktop #62: Wheel drop sensor data not exposed in simulation

ipa-kut, 2020-09-16:
I use the melodic branch of this repo, and start up the simulation with `roslaunch kobuki_gazebo kobuki_playground.launch`.   
I get the topics for the events for the bumper and cliff sensors under the namespace `/mobile_base/events`, but not the wheel drop sensors.   

Is there any easy way I can expose this data via a topic as well?   
Else, if I need to write some source code to make this happen, could I please be pointed to the right files to edit? This would save me the many hours required to study the architecture of this stack and trace the origin point.

corot, 2020-09-17:
Wheel drop sensors are not simulated. I suppose you can implement them easily by checking if motor wheel links are in contact with anything other than the robot base. U can do so in the plugin: https://github.com/corot/kobuki_desktop/tree/melodic/kobuki_gazebo_plugins and use bumpers as reference

ipa-kut, 2020-09-17:
> 
> 
> Wheel drop sensors are not simulated. I suppose you can implement them easily by checking if motor wheel links are in contact with anything other than the robot base. U can do so in the plugin: https://github.com/corot/kobuki_desktop/tree/melodic/kobuki_gazebo_plugins
> and use bumpers as reference

Thank you for the suggestion! I am not sure yet if I want to do this, but I will update this post if I do.

ipa-kut, 2020-09-28:
I have added contact sensors to the wheel_left link based closely on the bumpers on base_link : [here](https://github.com/ipa-kut/kobuki/blob/kut/feature/wheel_drop/kobuki_description/urdf/kobuki_gazebo.urdf.xacro#L12)   
I have also added ros plugins to publish these on topics for both the contact links : [here](https://github.com/ipa-kut/kobuki/blob/kut/feature/wheel_drop/kobuki_description/urdf/kobuki_gazebo.urdf.xacro#L19)   
Furthermore, I added a name to the collision tag of the wheel_left link to closely match base_link : [here](https://github.com/ipa-kut/kobuki/blob/kut/feature/wheel_drop/kobuki_description/urdf/kobuki.urdf.xacro#L76)  

What I now observe is that the topic published by the ros plugin of the base contact sensor  (/bumpers_contact) shows some good values. but the wheel_left contact sensor (/wheel_left_contact) does not. 

Furthermore, these wheel_left contacts are also not picked up in the C++ code [here](https://github.com/yujinrobot/kobuki_desktop/blob/melodic/kobuki_gazebo_plugins/src/gazebo_ros_kobuki_updates.cpp#L339). I expect the contact_size() to show counts for all contacts of both the base link and wheel left link, but only the base link contacts are counted. I may also not have understood this part correctly.

@corot , could you please suggest where I could fix this or what I could try instead?

ipa-kut, 2020-09-28:
> I have added contact sensors to the wheel_left link based closely on the bumpers on base_link : [...]

I figured out that the collision name that I should give in the gazebo plugin should match the collision name generated in the sdf from the urdf (based on an answer from ROS answers that I can no longer recollect) I have since fixed [this](	        <collision>wheel_left_link_fixed_joint_lump__wheel_left_collision</collision>) line and now it works as expected!

corot, 2020-09-28:
Cool, thanks!  I'll give a try this week, but cannot tell when I'll have time.
But why do you need the libgazebo_ros_bumper.so? You can get the sensor readings within the kobuki base plugin and publish there wheel drop events, as happens with the bumpers. See the code on kobuki node for reference:
https://github.com/yujinrobot/kobuki/blob/melodic/kobuki_node/src/library/kobuki_ros.cpp#L318

ipa-kut, 2020-10-02:
Created two PRs to address this:   
https://github.com/yujinrobot/kobuki/pull/425    
https://github.com/yujinrobot/kobuki_desktop/pull/64   

I also have a version in my fork where I add some additional features that I need for my experiments:   
1. Wheel drop sensors for the caster wheels as well
2. ROS topic publisher plugins for the contact sensors (main wheels and casters)
3. Ground truth pose publisher plugin

If any of these features seem useful to the main repo, please let me know, and I can include them in the PRs as well.

corot, 2023-10-11:
Merged to melodic
Requires pulling https://github.com/yujinrobot/kobuki too

closed 2023-10-11 by corot